adamfk/bouncy-button-data #24: 187us - MX Keyswitch - Gateron G Red Pro 2.0

Uzmeyer, 2024-04-15:
Please note: 
- This issue is for a **particular instance** of a switch/button of the original poster. If someone else wants to test the same switch model, they should create a new issue.
- The original poster can add additional test sessions to this issue (if they wish) by adding a comment. Or they can add a new issue.

# Switch Info

Property                           | Description            
--------------------------- | -----------------------
Type                                 | push button
Function                           |  Off-Momentary (Normally Open)
Model info/markings       |  Gateron, clear top, white bottom/ PN: KS-9H10B045NN-Y35
Voltage & Current rating |  12V, 10mA

**Product URL:**  https://www.gateron.co/products/gateron-g-pro-2-0-switch-set?variant=40036010459225

**How new is the switch?** barely used but has not been used for 2-ish years.

**How do you like the switch?** One of my go-tos at a very affordable price.

**Anything else noteworthy?** Supposedly pre-lubed, however I did not open it to check.

**Switch Description/Photos**
[image]

<br><br>

# Test Session 1

## Setup

Property                                      | Description
----------------------------------- | -----------------------------------------------------
Arduino model                            | Uno
Pull-up resistor                            | Arduino internal
Switch wires connection              | alligator clips
Measure switch's ON resistance | 0.5 Ohm including wires
Anything else connected?           | No

**Test Setup Description/Photos:**
Quite janky with aligator clips + dupont wires but the 3d printed holder made it quite solid and easy to press consistently.
[image]

<br><br>

## Results
### 📄 Raw Data

[Gateron Red V2.txt](https://github.com/adamfk/bouncy-button-data/files/14982697/Gateron.Red.V2.txt)

You can graph/analyze the data [here.](https://adamfk.github.io/bouncy-button/bbb-browser/gui.html)

### 🖼️ Summary Image

[image]

### Test Section Details
All tests were done without a keycap, I doubt this has much of an inpact on the result but is not how these switches would be used.
Normal tests I tried to push the switch about how I would when typing. Fast I hammered the switch quite hard and then slid my finger off so it returns under it's own spring. Slow I conciously moved the stem, not a situation this kind of switch would normally encounter

### 🕵️ Your Observations

I expected very little bounce from how these switches are constructed but wow, that is really little bounce. Interestingly the switch seems to prefer faster actuation. My guess is that some bounce is caused by the stem wobbling, so pushing it past the transition zone faster is benificial.
There was one outlier, the only one that pushed past a ms but seeing how consistent the switch is otherwise this was probably a connection issue.

adamfk, 2024-04-15:
Wow! That is a great switch. Thanks for sharing!

This is the first switch tested in this repo that actually has built in "hysteresis". All other push type buttons that I've tested to so far will repeatedly trigger if I press them slowly (that's why I disable that section in the data). I think that's because as I get close to the contact point, slight vibrations in my finger cause it to connect/disconnect repeatedly until I push well past the transition point.

One note about comparing the fast vs normal sections. I've noticed in my own testing that switch bounce appears to be largely random. The more tests we run, the more likely we are to see a switch's worst behavior. To be able to say with some confidence that Fast presses did better than Normal presses, they should each have roughly the same number of tests (same opportunity for bad bounce). I'd like to pretend that I knew this all along, but I only recently realized this after my 20th session or so :)

Here's the data when I've limited Normal and Fast to 20 tests each:
[image]

Fast does a tiny bit worse than normal.

That said, the Slow presses definitely look like they bounce less. They don't really resemble any other sections at all.

Interesting shape/patterns when sorted by duration:
[image]

Uzmeyer, 2024-04-16:
Interesting, I honestly only looked at the max bounce times when making that observation, not the overall statistics. I think I'll test this switch a second time with equal amount of normal and fast/aggressive presses as I think those are relevant in a keyboard but might drop the slow ones for future switches for the same reason.