adamfk/bouncy-button-data #25: 279us - MX Keyswitch - Ajazz × Huano Diced Fruit Kiwi

Uzmeyer, 2024-05-07:
Please note: 
- This issue is for a **particular instance** of a switch/button of the original poster. If someone else wants to test the same switch model, they should create a new issue.
- The original poster can add additional test sessions to this issue (if they wish) by adding a comment. Or they can add a new issue.

# Switch Info

Property                           | Description            
--------------------------- | -----------------------
Type                                 | Push Button
Function                           |  Off-Momentary (Normally Open)
Model info/markings       |  Ajazz, Transparent green housing with gold glitter
Voltage & Current rating |  N/A, probably 12V, 10mA

**Product URL:**  Seems to have disappeared.

**How new is the switch?** Never used.

**How do you like the switch?** I currently use its lighter version in one of my keyboards and do enjoy the agressive bump and short travel. Also it looks really nice.

**Anything else noteworthy?** Supposedly pre-lubed, however I did not open it to check.

**Switch Description/Photos**

[image]
[image]

<br><br>

# Test Session 1

## Setup

Property                                      | Description
----------------------------------- | -----------------------------------------------------
Arduino model                            | TODO: Uno
Pull-up resistor                            | TODO: Arduino internal
Switch wires connection              | TODO: alligator clips
Measure switch's ON resistance | TODO: 0.5 Ohm
Anything else connected?           | TODO: No

**Test Setup Description/Photos:**

[image]

<br><br>

## Results
### 📄 Raw Data

[Ajazz Kiwi.txt](https://github.com/adamfk/bouncy-button-data/files/15238785/Ajazz.Kiwi.txt)

You can graph/analyze the data [here.](https://adamfk.github.io/bouncy-button/bbb-browser/gui.html)

### 🖼️ Summary Image

[image]

### Test Section Details

All tests were done with a keycap this time.
Normal tests I tried to push the switch about how I would when typing. Fast I hammered the switch quite hard and then slid my finger off so it returns under it's own spring. Slow was hard to do due to the aggressive tactile bump and are really not something that would be encountered in normal use.
Tests 157,139,133,129,107,103,89 were excluded due to being heavy outliers.

### 🕵️ Your Observations

I did an equal amount of normal and fast presses this time but i think i will omit fast in the future, unless i find a better way to hook up the switch, as I can't rule out alligator clips causing outliers. However, this switch seems to behave very consistently when actuated in the same way, except for slow presses, however it is also really hard to actuate consistent and slowly due to the agressive bump in the travel. I think I will also leave those out in future tests due to low relevance but do more normal presses instead.

adamfk, 2024-05-08:
Nice! I also only do a few slow presses, because I find them super tedious. Maybe 10. It will often let you know if the switch has hysteresis though.

The fast outliers don't look that bad. Most of these switches would be rated for < 5ms and 1.6ms is well within that. I would only worry about loose wires if the bounce was really long like above 20ms.

I actually just tested a bunch of alligator clips. A good set would never glitch even when I smacked/flicked/twisted the wire/alligator clip. My worst set of alligator clips glitched for 2 microseconds once out of 80 attempts. I had to use a ridiculous amount of force though.

If you are worried about your connections, set it all up and then hit the housing/holder without activating the switch. I'll be you a dirty Cherry Brown that the alligator clips don't glitch unless you bring out a hammer :)

adamfk, 2024-05-08:
Also, nice looking switch 😻