adamfk/bouncy-button-data #26: 144us - MX Keyswitch - Gateron Brown

Uzmeyer, 2024-05-07:
Please note: 
- This issue is for a **particular instance** of a switch/button of the original poster. If someone else wants to test the same switch model, they should create a new issue.
- The original poster can add additional test sessions to this issue (if they wish) by adding a comment. Or they can add a new issue.

# Switch Info

Property                           | Description            
--------------------------- | -----------------------
Type                                 | TODO: push button
Function                           |  TODO: Off-Momentary (Normally Open)
Model info/markings       |  TODO: Gateron, clear top, black bottom/Model Nr. KS-8
Voltage & Current rating |  TODO: N/A, probably 12V, 10mA

**Product URL:** https://www.gateron.co/products/gateron-ks-8-mechanical-switch-set?variant=40089466601561
**How new is the switch?** TODO: Barley used, 4 - 5 years old

**How do you like the switch?** Used in my first custom keyboard, back then I really liked them but newer alternatives are simply better. Very wobbly stem and quite weak tactile bump. Still better than the original Cherry Brown.

**Switch Description/Photos**

[image]
[image]

<br><br>

# Test Session 1

## Setup

Property                                      | Description
----------------------------------- | -----------------------------------------------------
Arduino model                            | Uno
Pull-up resistor                            | Arduino internal
Switch wires connection              | alligator clips
Measure switch's ON resistance | 15 Ohm but jumps as high as 300
Anything else connected?           | oscilloscope? capacitor?

**Test Setup Description/Photos:**

[image]

<br><br>

## Results
### 📄 Raw Data

[Gateron Brown.txt](https://github.com/adamfk/bouncy-button-data/files/15239347/Gateron.Brown.txt)

You can graph/analyze the data [here.](https://adamfk.github.io/bouncy-button/bbb-browser/gui.html)

### 🖼️ Summary Image

[image]

### Test Section Details

Standard Cherry keycap used, trying to actuate the switch as when typing

### 🕵️ Your Observations

Not the most consistent switch, could be due to the tactile profile but very low overall bounce time despite relatively high on resistance indicating not the best connection.

adamfk, 2024-05-08:
Wow. Surprisingly little bounce considering the high resistance.

adamfk, 2024-05-08:
Also, I uploaded a new Arduino sketch https://github.com/adamfk/bouncy-button/releases

The older software would print out a lot of blank lines when starting which is nice when quickly messing around with the software, but makes the .txt files look blank because there are 50 new lines at the start.

It also animates the start up and menu text a bit. Feel free to use whatever you like best.